gin66/FastAccelStepper #337: forceStopAndNewPosition(0) generates an error E (25451) rmt: rmt_tx_disable(842): channel can't be disabled in state 0

Jay-esp, 2025-09-04:
ESP32, arduino IDE, using forceStopAndNewPosition(0) works but generates this error on the serial monitor:
E (25451) rmt: rmt_tx_disable(842): channel can't be disabled in state 0

gin66, 2025-09-04:
This error states (according to grok), that the rmt is disabled, while the rmt is in idle state.

The espidf version is 5.x ?

When `forceStopAndNewPosition(0)` is called, is the motor active? Or are there two or more motors all stopped at the same time - independent of the running state? The current forceStop algorithm does not test if the motor is actually running. This could be changed, but makes only sense if it was actually called on a stopped motor.

gin66, 2025-09-07:
The latest version avoids rmt_disable, if the channel is not yet enabled.
Please try