gin66/FastAccelStepper #338: As Fluidnc ESP32 I2S ouput  can out add this mode

Speedjf37, 2025-09-13:
Hello ,

FastAccelStepper is ok for ESP32 with GPIO ouput !

As Fluidnc I2S ouput  can out add this mode ?

    const EnumItem stepTypes[] = { { Stepping::TIMED, "Timed" },
                                   { Stepping::RMT_ENGINE, "RMT" },
                                   { Stepping::I2S_STATIC, "I2S_STATIC" },
                                   { Stepping::I2S_STREAM, "I2S_STREAM" },
                                   EnumItem(Stepping::RMT_ENGINE) };

Good job !
Thanks.
JF

soulslider, 2025-09-14:
Hello @gin66 and @Speedjf37 , 

@gin66 thank you for this uerfull library !

This is a great idea for driving cards like [MKS-DLC32](https://github.com/makerbase-mks/MKS-DLC32) where steppers are already wired to i2s output.

gin66, 2025-09-14:
Any port, which can be used as I2S-output can be instead connected to rmt-module, too. Thanks to esp32’s gpio- and io-matrix functionality.

soulslider, 2025-09-14:
Yes but on MKS-DLC32 boards all (step, dir, enable) pins for the 3 steppers are wired to the same esp32 i2s bus

[image]

gin66, 2025-09-15:
Thanks for this detailed info. That IC demultiplexes the I2S-stream and in the individual bits are the respective signals for the stepper. This means, one I2S would be able to manage multiple steppers. There would be the need to have one generic long I2S-buffer. Every stepper would need to update its respective signals’ bits for the whole buffer based on their individual timing. The bytes need to be ready before the DMA pulls the data in.

Somewhere I read BCLK is max 10MHz and 8bits minimum width, so each byte would represent 1.25us. For high speed operation not really good resolution. Now sure if this BCLK min. frequency is correct.